mletenay/home-assistant-goodwe-inverter #379: Extreme values since last update

Brkie, 2025-08-19:
Since the latest update of experimental, I’m getting extreme values for produced solar power, battery consumption, and battery charging. This happens almost daily. In SEMS everything looks normal, so I don’t understand where these values are coming from.

It happens randomly and can be corrected in the statistics, but it’s starting to get tiresome.

[image]

I have GoodWe GW10KN-ET with 13.3kW battery.

Nightenom, 2026-09-27:
Just here to amend and willing to do anything to fix this (have to fix values in statistics weekly..)
I have noticed that everytime this happens the time of the event is `hh:mm:00` (possibly even `hh:00:00`)

MildaC, 2026-09-28:
Apparently, this happens because the inverter returns a distorted value in response to an integration request. These are likely entities stored in the inverter’s EPROM. I filtered out these values in the auxiliary entities. However, after updating the inverter’s firmware to version 1333 and now to 1334, everything is working fine for me.
```python
      - name: "Meter Total Energy Export NEW"
        unique_id: meter_total_energy_export_new
        unit_of_measurement: "kWh"
        device_class: energy
        state_class: total_increasing
        availability: "{{ states('sensor.meter_total_energy_export') not in ['unknown','unavailable'] }}"
        state: >
          {% set current = states('sensor.meter_total_energy_export') | float(0) %}
          {% set previous_state = states('sensor.meter_total_energy_export_new') %}
          {% if previous_state in ['unknown', 'unavailable'] %}
            {{ current }}
          {% else %}
            {% set previous = previous_state | float(0) %}
            {% if (current - previous) | abs > 100 %}
              {{ previous }}
            {% else %}
              {{ current }}
            {% endif %}
          {% endif %}
```